tloimu/adapt-ffb-joy #19: Circuit Diagram in HowToBuild.md not found

dlwalter, 2018-02-14:
Clicking on the link for the circuit diagram goes to a 'not found' page.

http://adapt-ffb-joy.googlecode.com/files/adaptffbjoy-circuit.png

Located on https://github.com/tloimu/adapt-ffb-joy/blob/wiki/HowToBuild.md

schmatzler, 2018-03-02:
Here's the file in case anyone needs it:

[image]

schmatzler, 2018-03-02:
@Poil restored all of the missing images in his fork:

https://github.com/Poil/adapt-ffb-joy/commit/0f6614bbac89cbe4b5a01f06ecb1e401c46e1fb3

tloimu, 2018-09-12:
Fixed the image link and a couple of other similar. The problem was the Google Code's port to Git that it didn't take into account the links withing the site.

closed 2018-09-12 by tloimu