tloimu/adapt-ffb-joy #20: Not work on ATmega32u4 on Arduino pro micro Leonardo Board

denisn73, 2019-03-20:
I flash chip with "adaptffbjoy-r54.hex" and it not work! Device not recognized, no device in device manager.
But arduino bootloader work perfectly, this means that the board is working.

edcastanha, 2019-03-21:
How many reset drives are running?

denisn73, 2019-03-22:
> How many reset drives are running?

what is meant?

Dobie007, 2020-01-04:
The device will not be recognized if line 3 to PB5 is not connected correctly. If only the FFB does not work (sometimes) try to invert X1 - that solved the issue for me. I changed the Hex code accordingly. Adding capacitors did not help, i even removed those on the Joystick mainboard.

lucaceriani, 2020-05-17:
> The device will not be recognized 
Do you mean that if i connect only the board without the joystick it doesn't even show up on windows?

Dobie007, 2020-05-17:
No, sorry i was not precise enough: the joystick is recognized and works but there is no force feedback (exept in seldom lucky cases). Since the change it works perfectly-

HotteX, 2020-06-25:
>  If only the FFB does not work (sometimes) try to invert the X1 line 11 to PB4 - that solved the issue for me

Help me out please. What do you mean by invert? Swap PB4 and 5 ? I seem to have that issue but I am a total noob when it comes to coding.

Dobie007, 2020-06-26:
Invert means turn a high level to a low level and vice versa. You can either change the Hex-code or use a transistor to do so.

HotteX, 2020-06-26:
ah ok, that was the other option :D I'll give it a try right away, thanks!
Do you know where in the code the setting is hidden? I have started to make some other changes in the code that I would like to combine

HotteX, 2020-06-26:
tried the hex....could you check if that was the right one? It behaves like a sample-hex. its not detected as Lufa JoystickwFFB and gives a periodic push upwards on the stick every second.

Dobie007, 2020-06-26:
That sound wrong - i will need to search...

HotteX, 2020-06-27:
I didn't even notice that the Version 54 is different from the Git-sources. I didn't find any Version besides Lufa & 3DP-Vert...where did you find the code version?

One thing I notice is, that your hex is a bit bigger than the non-inverted one. 

r54 39.348 Bytes
r54_inverted 40.243 Bytes

Interesting thing: The guys over at decentbb took the code a bit farther it seems, integrating the Shift and more buttons. But they worked with the sources provided.

 Hexes & Text here https://dbb.d3k.org/viewtopic.php?f=8&p=386319
Sources here https://github.com/r8dhex/adapt-ffb-joy

Dobie007, 2020-06-27:
The original 54 file is in the download section of this repository: https://github.com/tloimu/adapt-ffb-joy/tree/master/downloads
 If i re download and unpack my inverted version it has exactly the same size - the packed version is a little bigger ..

HotteX, 2020-06-29:
I used 7Zip, thanks for the md5, I'll check my file. 
Thats where I got the v54 from, I just could not find any Version number in the GIT-Sources, thats why I thought that the v54-File is the result of the sources. 

Thanks for the help with finding the line in code. I'll give it a try in the evening.

HotteX, 2020-08-30:
I am a bit embarrassed to tell you (but also very happy).

I found the issue with my stick loosing force feedback while playing. Inside the joystick, a pin of the power connector was broken. So usually the stick had power and worked, but the vibrations of the motors made it loose it's power for FF. Replugging usually worked, because it pushed the pin back in place. 
I could beat myself with the stick for not even considering that the vibrations might have caused a physical problem. ARGH! 

After 30 seconds of soldering the beast is working.

drepou, 2021-03-08:
same kind of isssue for me on arduino micro stick is well recognised all axis ok on joystick conf i feel some motor resitance but not effect works with dcs game or force test programe any advice ? i tried the inverted Hex joystick is not reconized at all with this one

Hedda, 2023-04-17:
Anyone managed to port/patch latest adapt-ffb-joy to run on standard "Arduino Pro Micro" and/or "Arduino Leonardo" boards? 

According to the wiki at https://github.com/tloimu/adapt-ffb-joy/wiki there was an old project making it into an Arduino library:

https://code.google.com/archive/p/sidewinder-arduino/

ej113, 2023-04-27:
Yes!

I was going to try changing the pinouts to make it compatible myself but I instead stumbled across [this fork](https://github.com/juchong/adapt-ffb-joy) by @juchong which swaps PB0 for PB6.

It also has a PCB design, flashing instructions and is forked from the JayBee-git version with the shift button function and no extra axes.
This one worked for me first time yesterday on the Arduino Micro (not Pro Micro) and I would expect it to also work on the Arduino Leonardo (although using the ICSP header pins) and Adafruit Itsy Bitsy 32u4 5V 16MHz variant based on broken out pins. LEDs are still assigned to Teensy pins so don't turn on but this is a minor change if required.

Also the original Teensy 2.0 .hex files using the PB0 pin _should_ work with the Arduino Micro with no modification and with the Pro Micro if you solder to the LED pad. 

- See [here](https://www.descentbb.net/viewtopic.php?p=345065#p345065) for the Arduino Micro.  I was not able to get the FFB to work but it was most likely my dodgy breadboard wiring. Note the board RX LED is still connected to the PB0 pin. I wasn't sure if this might affect the function.
- and here for the Pro Micro #35

MacGH23, 2024-03-03:
I just tried to get the Micro Pro working.
In general is is working.
The joystick is connected and all Axis and Buttons working, but it is initialized with
FFB effects on and center it. The FFBtest program don't work.
Like https://github.com/tloimu/adapt-ffb-joy/issues/20#issuecomment-793078155
The teensy 2.0 I also have is working without problems.

ej113, 2024-04-06:
Sounds similar to #56. Maybe capacitors?
I've just added a [troubleshooting guide](https://github.com/tloimu/adapt-ffb-joy/blob/wiki/Troubleshooting.md) to the Wiki. Have a look and see if anything there works for you.

MacGH23, 2024-04-13:
I try 1nF upto 10nF in small steps.
Nothing worked.

Zyzyme, 2024-04-17:
Report: I have two joysticks #96755 device (no fan) and 96755 device (with fan)，

MacGH23, 2024-04-17:
Is this the right schematics for it ?
https://github.com/juchong/adapt-ffb-joy/tree/master/pcb

Zyzyme, 2024-04-17:
> Is this the right schematics for it ? https://github.com/juchong/adapt-ffb-joy/tree/master/pcb

I have Teensy 2.0 and Pro Micro working fine on v0.5.0beta1
I made a picture, you can check it out
[image]

MacGH23, 2024-04-17:
Thanks. 
PD0 and PB6 are connected - right (Pin 3 and 10) ?

Zyzyme, 2024-04-17:
> Thanks. PD0 and PB6 are connected - right (Pin 3 and 10) ?

Yes (PIN3 and PIN10)
https://golem.hu/article/pro-micro-pinout/

MacGH23, 2024-04-17:
Thanks, Then I check all again

MacGH23, 2024-05-23:
I found the problem.
The model of the arduino in my CAD program was wrong and it mixed the wires to the wrong pins.
I made a new PCB and now it is working, but only 20%.
Need to replug many times before it is working.
I try different capacitors the next weeks