tloimu/adapt-ffb-joy #25: Linux Force Feedback Support?

eholk, 2020-04-25:
I built one of these adapters for my Force Feedback Pro. Under Windows, it seems to work fine, both the joystick and the force feedback effects. Under Linux, however, I have only been able to get the joystick working. Force feedback does not seem to work. Even if I can't get full force feedback support working, it'd be nice to be able to turn on auto-centering.

I'm guessing this is more an issue with the Linux force feedback drivers, since adapt-ffb-joy presents itself as a standard USB HID device, but I figured I'd raise the issue here in case other people are having the same issue.

Running `fftest`, I get the following:
```
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/by-id/usb-Dean_Camera_LUFA_Joystick_wFFB-event-joystick opened
Features:
  * Absolute axes: X, Y, Z, RX, RY, RZ, Throttle, Hat 0 X, Hat 0 Y, 
    [7F 00 03 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: 
    Force feedback periodic effects: 
    [00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ]
  * Number of simultaneous effects: 0

Uploading effect #0 (Periodic sinusoidal) ... Error:: Function not implemented
Uploading effect #1 (Constant) ... Error: Function not implemented
Uploading effect #2 (Spring) ... Error: Function not implemented
Uploading effect #3 (Damper) ... Error: Function not implemented
Uploading effect #4 (Strong rumble, with heavy motor) ... Error: Function not implemented
Uploading effect #5 (Weak rumble, with light motor) ... Error: Function not implemented
Enter effect number, -1 to exit
```

When I look at a list of [supported devices](https://github.com/Eliasvan/Linux-Force-Feedback/wiki/Supported-Devices) under Linux, it looks like these are all hardware specific, rather than a generic USB device. I don't know how accurate this list is. I haven't been able to find anything that specifically mentions Linux support for USB HID force feedback.

Even if Linux doesn't support this, I've seen some mentions on this repo of a serial port that can be used to configure the adapter. Is it possible to use this to at least turn on auto-centering mode?

tloimu, 2022-04-25:
Looks like Linux PID-drivers work differently than Windows drivers at least with this stick. The USB-side descriptor and data model of this adapter is actually derived from the MS FFB 2's USB-descriptor and data model. At least the above effects 0, 1 and 2 are implemented in the adapter and work under Windows, so Linux clearly works them differently.

If Linux is still of an interest, maybe someone can fork this project and develop it there. Closing the issue here.

closed 2022-04-25 by tloimu

ej113, 2023-10-28:
[This thread](https://news.ycombinator.com/item?id=28816377) suggests that generic PID descriptors (without custom drivers) compatible with Windows will not be compatible with Linux and vice versa.