tloimu/adapt-ffb-joy #26: no gas slider in X-wing  Alliance game

gboule65, 2020-07-28:
I think is not a real bug, but I saw many people with is bad behaviour.
This old game use "rudder" HID definition instead of "throttle" HID, and this game don't permit to change the slidder assignment.

So, I have modified the source code in order to have the game functional :
  - HID definition modification : rudder only instead of rudder & throttle
  - remove of Z, Rx, Ry, 
  - joystick slider data send to rudder instead of throttle

I can give the source code modified.

Thank you to the different teams about the great adapter which permit me to play with my old MS Sidewinder Force Feedback Pro.

galvanized, 2020-08-16:
Could you upload your fork? Interested in what you did.

gboule65, 2020-08-16:
I’m on holidays now but I can upload next week (if I don’t forget...)