tloimu/adapt-ffb-joy #27: Throttle axis reversed, truncated

cbhacking, 2020-09-20:
FFB Pro, Win10. The throttle slider is marked on the case as increasing with clockwise rotation (this is also just what you'd expect; throttle forward should be "more"). However, the throttle axis shows as inverted in Windows; max clockwise (forward) is -64, and counter-clockwise (back) is +63,

I can correct for this in games that allow inverting the throttle axis, but it seems like it would be better to do that in the microcontroller.

Additionally, the throttle range seems to be truncated; roughly the top eighth and bottom eighth (so, 1/4 of the total movement range) of the controller doesn't affect the output at all. Obviously this could be due to things like the joystick being decades old or possibly it was always like that, but it seems a shame to waste that much of the throttle's movement range (and also makes minute corrections harder within the remaining range).

It's possible I've got some wires crossed somewhere on my breadboard, but I don't think so; I was pretty careful and double-checked.

retroluxfilm, 2024-01-10:
I can confirm that I have the same issue that not the full range of the throttle slider is used and that it is inverted. Would be great if that can be fixed at some point.

ej113, 2024-01-11:
The range of the throttle slider is limited by the joystick itself. The adapter only forwards the 7 bit value it receives.  63 appears to line up with the smallest dot and -64 with the largest dot. I just tested also on my USB Sidewinder Precision Pro (with the adapter that came in the original box) and it has exactly the same behaviour, including in fact the axis being reversed.

retroluxfilm, 2024-01-12:
Ohh thats good to know that this is the expected behaivor. So then everthing is in order and this is not exactly an bug per se. My Sidewinder has exatly the same raw value limits on the slider and range on the dots. Do you think it is  possible that there is some adjustment screw (poti) on the board to extend the translation of the movement and output value. The board has some adjustable potis but i dont know for what they are.