tloimu/adapt-ffb-joy #32: Optional pots can't be disabled in some games (eg IWAR2)

Leum-Dunn, 2021-02-20:
I bought an adapter with the optional pots installed and it works well.
I play space sim games like IWAR2 and the 'throttle' on the joystick is not recognised - instead the 2 'slider' pots are the throttle.
This cannot be configured in game, so I removed the pots from my adapter - but I'm guessing they need disabling in firmware since the game still has no throttle.
Can anyone help to make a firmware that only enables the standard joystick features?
Thank you in advance.

isopix, 2021-02-20:
There are different adapters(using different boards and possibly
different pinout layouts) and  customized firmware. You should contact
person that sold you the board or trace it on the boards and post the
result there or even correct firmware yourself(it's not hard to change
pinout layoud and compile it). It's even possible that what product
you bought is buggy, so you should check it in some software(control
panel o windows or some jstest under linux), because game you are
playing can be buggy too. I'm not an electronic, but by desoldering
pots, you've changed this input to floating gate, so it's askijg for
problems(am I right?)

YARD2, 2021-02-27:
You can try this cleanup FW
https://github.com/JayBee-git/adapt-ffb-joy
He had a similar issue with another game

Leum-Dunn, 2021-03-08:
> You can try this cleanup FW
> https://github.com/JayBee-git/adapt-ffb-joy
> He had a similar issue with another game

Thanks for this - I'll do some reading and learn how to flash the updated firmware.
If you have any links or recommended tutorial reading then please share.

YARD2, 2021-03-08:
Just use the teensy Loader
https://www.pjrc.com/teensy/loader.html

connect, press the button and flash it.
Quite easy

Leum-Dunn, 2021-03-08:
> Just use the teensy Loader
> https://www.pjrc.com/teensy/loader.html
> 
> connect, press the button and flash it.
> Quite easy

Worked brilliantly - I'm super happy - thank you very much :)

closed 2021-03-08 by Leum-Dunn