tloimu/adapt-ffb-joy #33: feature request: support start delay parameter

jmriego, 2021-02-22:
would it be possible to add support for the start delay parameter?

this parameter makes the effect start a certain number of milliseconds after the effect is played. This is used for example in the [ForceTest.exe](https://www.dirks-software.ca/downloads/ForceTest.exe) test script for playing the grooves effect which is basically feeling a bump with a certain frequency

ej113, 2023-08-19:
Both StartDelay and LoopCount parameters would be needed in order to implement this effect. The FFP doesn't appear to support either - I haven't found any trace of them and the [API documentation](https://archive.org/details/sidewinder-force-feedback-sdk-programmers-reference) also indicates they are not supported.

The only way this could be implemented is in the adapter itself. It would need to send repeat "Start Effect" messages to the joystick with a timed interval between them and also track the number of times the effect had played and whether the effect was supposed to be stopped.

This grooves effect is implemented in FS Force. However Dirk also provided a configurable software alternative that sends a repeat "Start Effect" since the Saitek joysticks don't appear to support StartDelay either.

jmriego, 2023-08-21:
thanks for your reply! I ended up creating a different repo https://github.com/jmriego/Fino for using an Arduino as a FFB and included the code there if you want to refer to it. The most important file for you would be [Joystick.cpp](https://github.com/jmriego/Fino/blob/master/src/Joystick.cpp)

I'm fine to close this ticket unless you want to keep it open for discussion. Thanks!

ej113, 2023-08-22:
Neat, that could be handy. I would leave it open since it is still a potential feature, even if not supported originally by FFP.