tloimu/adapt-ffb-joy #35: Need help for arduino pro micro schematic and connection

drepou, 2021-03-08:
Hi is there anyone who make work the .hex on arduino pro micro ? the pinout is not the same on the board so just "move" the connection on same name beetween the two boards but is it enough ? i dont have the PB0 on the pro micro board so i didnt make the PB0-PD0 connection and PD0 look to have two different places pin3 and pin14  dont know wich one use : 
https://cdn.sparkfun.com/datasheets/Dev/Arduino/Boards/ProMicro16MHzv1.pdf

i try .hex on my board flash is ok but after no joystick detection may be the wirering is bad or hex not compatible ? any help is welcome.   I use the old ffb1 sidewinder 96755 i also noticed that if i just plug the 12v adapter on the stick but no connected the gameport of the joystick i have no reaction of the stick no selfcentering and no resistance dont know if this is the normal behaviour or a faulty stick.... i was thinking that plug the power will center the stick even if not connected to a PC can someone do the test ?

drepou, 2021-03-09:
solution found please see my  auto comment

closed 2021-03-09 by drepou

shearn89, 2023-06-04:
Just to say a massive thank you for this! Was tearing my hair out with @juchong's fork thinking it would just work - but on my Pro Micro USB C version, the pin for PD0 was also off by one - had to be pin 3 or 10 as labelled on the pro micro, due to the fact they're connected. I now have a working joystick!!

juchong, 2023-06-09:
> 

No problem! I'm glad to hear my fork has helped others!

shearn89, 2023-06-13:
@juchong - while I have you here - do you know if the 'shake' effects on e.g. aircraft stall work with your fork? I have a Sidewinder Force Feedback Pro (1, not 2), and get no stall effects or ground rumble in DCS. Any ideas? I've tried with a force feedback test app and the 'sine wave' effect does nothing...

ej113, 2023-06-13:
Sine waves work for me. I tested with fedit and also with FS-Force in FSX. I could recognise nearly all of the effects I was expecting from the FS-Force documentation. However I did observe that any effect with less than about 40% gain was imperceptible, so I had to dial up most of the effect gains in software, otherwise they were too subtle. Might just be because my FFBP needs a service though. This wouldn't explain why you are getting no response in the test app however. Are you sure you're getting any effects other than the default spring centering?

shearn89, 2023-06-14:
I'll give it another try this evening with various apps. In the test app I was using I was able to see constant force, return to centre, damping - all worked fine. But sine wave didn't. Also in game I don't get any shake effects but I've heard reports that the FFB Pro 2 does... I'll try and do some more digging though.

Do you have DCS World (it's free for the base game)? If so, can you hop in the free TF-51D and let me know how it feels?

I guess it might be down to the game, not the firmware!

shearn89, 2023-06-14:
Okay - interestingly, this evening using some ForceTest.exe - none of the effects take hold. The stick is stuck in 'spring center' mode, when the other day I could get most of the options like 'constant force' to work and move the stick. I've also tried FEdit and set up a simple sine wave - no luck eiher! I'm planning on converting my breadboard setup to a more permanent protoboard setup, so will report back then. It may be some dodgy connections on the breadboard...

MacGH23, 2024-06-02:
Perhaps you need a bigger cap like me.
See
https://github.com/tloimu/adapt-ffb-joy/issues/20#issuecomment-2126491248