tloimu/adapt-ffb-joy #37: Its not really an "Issue" but rather me trying to get smart people to help me get this working with a full sized arduino...

ZeroTasteGuy, 2022-02-27:
Hi there, i have been trying to get my dads old sidewinder ff pro working again for a while now and have yet to be successful... i bought an gameport to usb adapter but its not "working" and while i was trying to troubleshoot that i stumbled across this lovely project. Now after reading through it i wanted to see if some of the "smarter" people here could help me get it working on my hardware

ZeroTasteGuy, 2022-02-27:
https://cdn.discordapp.com/attachments/750058894188871702/947456688255029268/IMG_20220227_123327.jpg
https://cdn.discordapp.com/attachments/750058894188871702/947456688628318238/IMG_20220227_123237.jpg

ZeroTasteGuy, 2022-02-27:
this is the part of the adapter i plan on using

aleraco, 2022-02-27:
Aleraco@gmail.com

ZeroTasteGuy, 2022-02-27:
> [Aleraco@gmail.com](mailto:Aleraco@gmail.com)

what?

aleraco, 2022-02-27:
I can give you help if you give me you mail

ZeroTasteGuy, 2022-02-27:
> I can give you help if you give me you mail

zerotasteguy@gmail.com

tloimu, 2022-04-25:
Moved to discussion side as https://github.com/tloimu/adapt-ffb-joy/discussions/38

closed 2022-04-25 by tloimu