tloimu/adapt-ffb-joy #44: arduino leonardo flashed with ffbjoy doesnt get recognized as a joystick

xaspar30, 2023-06-13:
i wired everthing up according to [https://github.com/juchong/adapt-ffb-joy](url) all the steps in there but i cant seem to get it recognized. (the joystick itself works fine)

tloimu, 2023-10-03:
The hex file here was compiled targeting Teensy 2 board. I have not tested it with anything else as that is the only atmega32u4 board I have. You may need to recompile the project from source targeting Leonardo instead.

BrianTheBrain11, 2024-04-03:
using the newest release 0.5beta, the micro file that lists the leonardo as possibly compatible. uploaded micro hex to arduino and joystick is not being recognized.

Hermesaile, 2024-04-04:
I use the Teensy 2.0 and i have the same problem, windows doesnt recongnize as joystick or anything. I use de Teensy Loader and i tried the (r54).hex and 0.5beta and nothing. 
Its an secretly step before de loader?

ej113, 2024-04-06:
I've just added a [troubleshooting guide](https://github.com/tloimu/adapt-ffb-joy/blob/wiki/Troubleshooting.md) to the Wiki. Have a look and see if anything there works for you.

Hermesaile, 2024-04-07:
Ey Ej113! Many thanks it was a problem with de capacitors!

Great job!

BrianTheBrain11, 2024-04-14:
My problem was that I did not realize controller needed to be plugged in to be detected by windows. that fixed my issue