tloimu/adapt-ffb-joy #53: Centering spring but no FF on 0.5beta

gufo2k20, 2023-11-10:
Teensy board, working good with previous release 0.3.0.
Upgrading to 0.5beta joystick correctly recognized as MS FFB Pro but only centering spring work, no FF at all.
Same circuit (no trimmers)

ej113, 2023-11-11:
One of the changes is that the centering spring is not removed on force feedback initialisation. (My reasoning was that it makes it easier to play games without force feedback). Games with FF will remove it before they start playing their effects.

Questions and things to try:
- Which hex are you trying? Is it the shift_4axis_Teensy one?
- Does it get recognised in windows and behave as a normal non force feedback joystick with buttons and axes functioning?
- If you open Force Editor (fedit.exe) the one that was included in old DirectX, does it warn you that it can't find a force feedback device? Do any of the effects play?
- does it work in any games or other force testing applications?

I will check later I haven't mixed up the hex files or something daft. I can't test the Teensy targeted ones directly since I don't have the board. I can only test on the Micro then remove the changes for the different pinot.

ej113, 2023-11-11:
And another
- what does the board LED do when you first connect, and when you play effects?

gh0stless, 2023-11-11:
The shift version runs well on my Teensy. Now the thrust control is also recognized by the old games. So far I have tested Force Edit, X2 and X3.

gufo2k20, 2023-11-11:
Hi, thanks for the answer.

By the way, i've just reflashed the Teensy to answer your question and looks FF now work in DCS and IL-2 Great Battles.

With ForceTest.exe i've got to do some reset and device selct but it works too (looks like a my problem with the GUI)

>  
>  
> 
>  
> 
> One of the changes is that the centering spring is not removed on force feedback initialisation. (My reasoning was that it makes it easier to play games without force feedback). Games with FF will remove it before they start playing their effects.

I agree with this solution

> 
> Questions and things to try:
> 
> * Which hex are you trying? Is it the shift_4axis_Teensy one?

All 3 hex. Teensy and Teensy_shift works equally good (even the shifter function of button 9)

Flashing Teensy with the "micro" hex not recognized by Windows (obviously, i think)

> * Does it get recognised in windows and behave as a normal non force feedback joystick with buttons and axes functioning?

With Teensy .hex, yes. All axis and buttons working, just the FF not working and joystick centering spring forces on.

> * If you open Force Editor (fedit.exe) the one that was included in old DirectX, does it warn you that it can't find a force feedback device? Do any of the effects play?

Sorry, don't have that executable to test on my Win10....

> * does it work in any games or other force testing applications?

yes, it was working good like a normal spring joystick

> 
> I will check later I haven't mixed up the hex files or something daft. I can't test the Teensy targeted ones directly since I don't have the board. I can only test on the Micro then remove the changes for the different pinot.

Let me know if i can help with Teensy board testing.

If i don't remeber wrong it's 2.0 with the schematics from your GiT, without the trimmer.

gufo2k20, 2023-11-11:
>  
>  
> 
>  
> 
> And another
> 
> * what does the board LED do when you first connect, and when you play effects?

LED stable on.

Testing forces, playing games with or without FF.

gufo2k20, 2023-11-11:
Hi Andreas, just replied to ej113.

Reflashed my Teensy, now looks it works correctly, shift or non_shift .hex....don't know why the  first two or three  times i've tried the hex, just the centering spring forces aws working...

gufo2k20, 2023-11-11:
so, after a few tries....looks like USB port related.....i'm not sure but FF work only on the "powered" ones of my old USB-hub.
Or maybe it's a Win's drivers problem

ej113, 2023-11-11:
> Il 11/11/2023 09:56 CET ej113 ***@***.***> ha scritto: And another * what does the board LED do when you first connect, and when you play effects?
> LED stable on. Testing forces, playing games with or without FF.

Now that's odd. Expected behaviour is
- Flash 3 times a second until connected to joystick
- solid on when connected to joystick
- turns off when connected to PC (the same time you hear the chime in windows)
- then flashes briefly whenever a force feedback message is received from PC

gufo2k20, 2023-11-11:
Oops...you mean the Teensy board! I got It inside a closure, sorry. Forget my answer about the led....i'll check later assp

gufo2k20, 2023-11-11:
> Now that's odd. Expected behaviour is
>
>     * Flash 3 times a second until connected to joystick
>
>     * solid on when connected to joystick
>
>     * turns off when connected to PC (the same time you [...]

So, i was referring to the FFB Pro led......cannot see any blinking led on the Teensy.
Tried with "blinking.hex" on teensy (i think it's a demo sketch from the Teensyloader), no led blinking or fix.....

ej113, 2023-11-28:
Closing this as it sounds like it was a host USB port issue and it is not clear if the missing LED might be an issue with the board itself.

closed 2023-11-28 by ej113