tloimu/adapt-ffb-joy #56: v0.5.0beta1: FFB effects are lost after USB adapter reconnect

retroluxfilm, 2024-01-28:
The beta works great apart that the FFB effects are not working anymore when reconnecing the (Teensy) from https://www.yard2usb.de/joomla/index.php/sidewinder-usb to usb again. It just works directly after programming the teensy. With the firware from https://github.com/JayBee-git/adapt-ffb-joy it works even when reconnecting the USB adapter. Not sure what is the issue with the beta.

The autocentering works but no effects are triggered by the ForceTest tool.

retroluxfilm, 2024-01-28:
Seems not be related to the beta only. Not sure what is the cause that the FFB are lost in general often after reconnection of the adaper.

ej113, 2024-01-28:
Are you sure it's just the FFB not working or is the adapter not being recognised at all by the USB host? Is reprogramming the firmware the only way to get it to reconnect?
 
I can reproduce something similar with the Arduino Micro that is maybe related? When I press the reset button on the Micro it enters the bootloader, then after a few seconds drops out of the boatloader and starts the adapt-ffb-joy firmware. Based on the board LED I can tell it gets as far as recognising the joystick, but then fails to connect to USB. I can sometimes also get this to happen by wiggling the USB connector so it loses comms but not power.

If the USB connector is fully removed and reinserted it initialises and connects normally.

I doubt there is anything in the beta firmware that would affect this behaviour - thanks for checking and changing the issue title.

retroluxfilm, 2024-01-28:
The adapter itself is recognized and you can play games with it etc. But no FFB effects apart from the auto centering. I still can not figure out what is causing it directly. ForceTest still recoqnises the available FFB effects but it seems it does not send it to the Joystick. Really odd and a hit and miss to get it back working. Mostly Reconnect the USB connection on and off again brings it back.

retroluxfilm, 2024-03-16:
Amazingly the author of the USB Adapter I use (https://www.yard2usb.de/joomla/index.php/sidewinder-usb) contacte me as he saw the issue I reported here. He found out that this issue is hardware related and that the two capacitors (1nF) on the board are to weak for some FFB devices. Replacing them with bigger capacitors (1.5nF) helped to fully resolve this issue for me.

closed 2024-03-16 by retroluxfilm

Michael-1357, 2024-03-29:
My adapter behaved in a very similar temperamental fashion.  After reading retroluxfilm's solution I tried higher value capacitors.  2.2nF resolved the problem for my board.   Thank you.