tloimu/adapt-ffb-joy #58: Pi Pico (RP2040) support?

mdr92, 2024-04-24:
I know it's currently not supported but has any investigation being done on porting it to a RP2040 device as they're cheap and readily available. I'm not a fan of the Pro Micro as find the micro usb ports rip off easily. Something like an RP2040-Zero with USB C would be great for this project, but many other RP2040 devices exist in various formats (lots with usb c).

https://www.waveshare.com/wiki/RP2040-Zero

ej113, 2024-04-25:
Well the RP2040 runs at 3.3V and is not 5V tolerant. It might work with level shifters which are also cheap and available but maybe the additional cost and complexity makes it not worth it.

As for the code it is reasonably modular but would still be a decent size project for someone to port it from LUFA to e.g. TinyUSB.  And Grendel's original code to read the Sidewinder joystick position packets is in AVR Assembly, although there is an alternative open source implementation in C++ under the Necroware project.

sonik-br, 2024-08-16:
Looks like the Pico 2 (RP2350) is 5v tolerant but the device must be powered. Not a problem if connected to usb first, then gameport wheel connected to it later.

"GPIOs are 5 V-tolerant (powered), and 3.3 V-failsafe (unpowered)"

NolanNicholson, 2024-11-11:
Hey there! I was also interested in an RP2040-based adapter for the joystick, so I went and did the thing.

https://github.com/NolanNicholson/picowinder

It is a new project inspired by info from adapt-ffb-joy, rather than a direct port of it. For example, there is no Sidewinder Force Feedback Wheel support. But Linux is supported!